bigtreetech/BTT-TFT35-E3-V3.0 #77: Y axis not homing in touch screen mode

Tktrouble, 2022-02-01:
Hi  I have been having problems with my screen I hope someone can help because I have tried big tree tech and I sill got no where.

In marlin mode and on my computer it home all axis fine, but when I use the touch screen the y axis won’t home. I have tried it on 2 different tft35 e3 v3 screens and tried changing the cords. same out come. 
Thanks for the help

marlin firmware is: 2.0.9.3 
Board is: BTT SKR PRO 1.2
Tft 35 e3 v3 firmware is form: January 25

Tktrouble, 2022-02-01:
Sorry not the best video of the problem 
https://user-images.githubusercontent.com/98793743/151911396-e0a20fe0-1e10-4a80-a23b-c52f3c6cc463.MOV
Thanks

radek8, 2022-02-01:
Marlin writes the error message, not the display.
You have a problem with the Marlin configuration

Tktrouble, 2022-02-01:
They why does it home when use marlin mode 
But doesn’t with the touch screen
If it is marlin what do you think I need to change in it

radek8, 2022-02-01:
send configuration

radek8, 2022-02-01:
What end stops do you use?

[image]

Tktrouble, 2022-02-01:
That happens when the Y Axis/ bed homing

radek8, 2022-02-01:
I understand
Submit those configuration files.

radek8, 2022-02-01:
Did I hear a click on the X-axis, microswitches?

radek8, 2022-02-01:
It is late I am going to sleep

Tktrouble, 2022-02-01:
Ok 
Ya I am using end stop microswitches
I will send the files when I get home thanks

Tktrouble, 2022-02-02:
here is my files
[Marlin-2.0.x skr pro1.2-tft35-bltuch.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/7983354/Marlin-2.0.x.skr.pro1.2-tft35-bltuch.zip)

Tktrouble, 2022-02-02:
here is my files
[Marlin-2.0.x ET5X.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/7983365/Marlin-2.0.x.ET5X.zip)
thanks for the help

radek8, 2022-02-02:
In Marlin you have set up
#define PREHEAT_BEFORE_LEVELING
Therefore, the Home function does not run unless the printer is preheated.
Isn't that your problem?
Try to comment on this option.

Tktrouble, 2022-02-02:
It might be 
But would that affect just for when it’s just homing not doing bed leaving

radek8, 2022-02-02:
I don't understand what you mean

radek8, 2022-02-03:
Then there may be a problem in Marlin.
Close your ticket here

closed 2022-02-03 by Tktrouble