bigtreetech/BTT-TFT35-E3-V3.0 #79: Filament load notification spam bug

PhilippK233, 2022-02-08:
After I press Filament load and the filament has been loaded the following notification appears: " **Printer is Paused**  Paused for user continue?" When I press ok this notification appears before I can do anything and nothing happens. And this happens till I turn off my printer. So I looked in marlin mode if it works there and: yes AND after it loaded the Filament there are two options: "Purge more" and "Continue". When I press "Continue", the screen just disappears and I can use my Printer as normal. If I press "Purge more", it extrudes more Filament and then the options appear again and I can choose again. I have two pictures attached my Marlin Firmware and my TFT config.

Here are my printer specs:

- BTT skr1.4 turbo
- BTT tmc 2209 (doesn´t really matter)
- BTT TFT 35 V3.0 E3

Thank you

[image]

[image]

[Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/8026980/Marlin-bugfix-2.0.x.zip)

[config_rrf.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/8026983/config_rrf.zip)

PhilippK233, 2022-02-11:
> Make sure Marlin has all necessary functions enabled which are mentioned in the ReadMe on the TFT firmware repository. 

That's the thing😂 I already looked for that, everything, which is mentioned, was already enabled😂

oldman4U, 2022-02-11:
And Marlin is version?

PhilippK233, 2022-02-13:
> And Marlin is version?

Bugfix-2.0.x

radek8, 2022-02-15:
> > Make sure Marlin has all necessary functions enabled which are mentioned in the ReadMe on the TFT firmware repository.
> 
> That's the thing😂 I already looked for that, everything, which is mentioned, was already enabled😂

It is not true.
You do not have the required parameters enabled.

you have to uncomment:
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT
#define M114_DETAIL     
#define REPORT_FAN_CHANGE 
#define M115_GEOMETRY_REPORT
#define AUTO_REPORT_POSITION
#define AUTO_REPORT_POSITION
#define EMERGENCY_PARSER
 #define AUTO_REPORT_SD_STATUS

Why did you enable this option?
   #define BLTOUCH_SET_5V_MODE
Your board has 3.3V logic.
5V can damage some pins.

Where did you connect the Probe? You do not have a Probe pin enabled