bigtreetech/BTT-TFT35-E3-V3.0 #80: My TFT35-E3 touch screen displays random characters on the Left edge and scramble the screen from time to time

chobson303, 2022-03-01:
I've been having problems with my Z offset not being remembered and problems with the display in Marlin (mostly), When I try to readjust the offset in the Marlin mode the screen will sometimes display extra characters on the right side of the screen and from time to time the screen gets scrambled.  I have the touchscreen installed on an Ender 3 pro and I've had it about a year and a half.  Trying to get some ideas way the screen is acting up mostly in Marlin and why it won't remember my Z offset.

carl1961, 2022-10-23:
Might be late, but this is a issue I see too. the display is a simulating marlin TFT so it seems to me a serial emulation issue.

LouDnl, 2023-10-14:
Issue is still present, have the same thing

LouDnl, 2023-10-14:
After reading what @TGTools79 described here https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/335#issuecomment-1427299884 I was able to resolve my issue with the random characters and the display freezing.

What I did was:
- Add a ferrite bead around the Pi4 power cable close to the USB-C input. 
  I previously had the Pi4 powered via printers power supply when the printer turned on and the issues appeared after I moved to an external power adapter.
- Reroute the 3 display cables around my Pi4 instead of over the Pi4.
- Add a ferrite bead around the display tft serial cable close to the mainboard
- Move the USB cable from a USB3 port to a USB2 port on the Pi4
  USB3 is known for causing interference at around 2.4Ghz (especially on GPS modules), see explanation here: https://www.usb.org/sites/default/files/327216.pdf
  this could have been causing some issue with the serial cable, dont know for sure but you never know ofcourse

Now it seems solved, will post here if not :)