bigtreetech/BTT-TFT35-E3-V3.0 #82: TFT70 Vertical GUI

Akds4877, 2022-05-30:
Hey, i want to place my TFT on vertical point and there is no space in my Printer body for horizontal, so i did the configuration in portrait mode but it does not look well as the orientation is not so cool so please help me to make my TFT70 on Vertical Mode.
[image]

radek8, 2022-05-31:
The vertical mode is still under development and is not fine-tuned.
Icons and Appearance is primarily designed for horizontal mode. Because the same icons are used, vertical mode will not be as perfect as horizontal. Nevertheless, the programmers try to adapt it as best they can.
This storage, however, unfortunately monitors few users.
It will be better if you publish your problem here.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues

Maybe someone will react to it.

Akds4877, 2022-06-01:
Okay 

Thank you

closed 2022-06-01 by Akds4877