bigtreetech/BTT-TFT35-E3-V3.0 #84: Getting error when compliing

XeroZar, 2022-10-26:
Hi guys, hope some knows a fix for this... When I compile the firmware, I get the follow 2 errors...
==============================================================
TFT\src\User\Hal\STM32_USB_OTG_Driver\src\usb_core.c: In function 'USB_OTG_WritePacket':
TFT\src\User\Hal\STM32_USB_OTG_Driver\src\usb_core.c:178:7: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  178 |       USB_OTG_WRITE_REG32( fifo, *((__packed uint32_t *)src) );
      |       ^~~~~~~~~~~~~~~~~~~
TFT\src\User\Hal\STM32_USB_OTG_Driver\src\usb_core.c: In function 'USB_OTG_ReadPacket':
TFT\src\User\Hal\STM32_USB_OTG_Driver\src\usb_core.c:204:5: warning: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Wattributes]
  204 |     *(__packed uint32_t *)dest = USB_OTG_READ_REG32(fifo);
==============================================================

Anyone have a fix for this... I've been looking for a while found suggestions but no fixes.

XeroZar, 2022-10-27:
Ok figured it out again. With no help. Please close this thread.

radek8, 2022-11-05:
The ticket must be closed by the person who opened it, i.e. you

closed 2022-11-07 by XeroZar