bigtreetech/BTT-TFT35-E3-V3.0 #85: Random freezes

Thomas-tuta, 2022-11-07:
I have an btt skr mini e3 v3 with marlin 2.1.1 and a tft35 e3 v3 running on my artillery sidewinder x1. When using the sdcard slot or usb port on the tft, i get random freezes during the print. i have tried many sdcards(all 8gb or less, formatted fat32) and many usbsticks and they all have the same problem. also when powercycling the printer with the sdcard or usb stick inserted the tft wont start, it only gives an btt boot logo. I have a current workaround of having a micro sdcard in the skr mini and printing from that. but my motherboard is located very difficult to use this micro sdcard slot. changing to the latest firmware on the tft did not help. using the micro sdcard with an adapter on the tft still gives the freezing problem.

XeroZar, 2022-11-07:
I have the same issue but with mine I use octoprint as well and it does the exact same thing. If I print and don't touch the screen it prints fine. But when I have to tune steps or speeds it freezes

radek8, 2022-11-08:
When reading various discussions, I think that the SKR mini E3 V3 board has a big problem with electromagnetic interference.

I would try urovant wiring. Use a shielded serial cable to connect SKR and TFT. Store the display and serial cable as far as possible from the cables that power the stepper motors.

radek8, 2022-11-09:
you can make it yourself from an old mouse cable

[image]

XeroZar, 2022-11-11:
Cool thanks will try... what's the connector called so I can order them?

radek8, 2022-11-11:
XH2.54

XeroZar, 2022-11-13:
Cool thanks.

XeroZar, 2022-11-13:
Do the TFT35 screen create some sort of debug log as to what they are doing when the get the Busy Error?

radek8, 2022-11-13:
Busy is not an error, it is a message from the printer that it is busy with some activity and is not accepting further commands.

radek8, 2022-11-13:
Yes, the serial communication debugging function is integrated.
It is disabled by default.
To activate it, you have to turn it on in the Configuration.h file and compile your own FW.

 * Generic Debug
 * Uncomment/Enable to enable arbitrary debug serial communication to SERIAL_DEBUG_PORT defined in board specific Pin_xx.h file.
 */
//#define DEBUG_SERIAL_GENERIC  // Default: commented (disabled)

/**
 * Serial Communication Debug
 * Uncomment/Enable to forward/dump all serial communication to SERIAL_DEBUG_PORT defined in board specific Pin_xx.h file.
 */
//#define DEBUG_SERIAL_COMM  // Default: commented (disabled)

/**
 * Config File Debug
 * Uncomment/Enable to show debug information during config file processing.
 */
//#define DEBUG_SERIAL_CONFIG  // Default: commented (disabled)

XeroZar, 2022-11-13:
So if the board freezes, will it generate and put the log on the sd card?

radek8, 2022-11-13:
I don't know. I have never tested this feature.

XeroZar, 2022-11-13:
Ok so the log will be on the sd card? If it generates one...

XeroZar, 2022-11-17:
> you can make it yourself from an old mouse cable
> 
> [image]

Ok so after researching that does not look like a shielded cable.

XeroZar, 2022-11-17:
Success, after many weeks I finally found the issue. No more freezing, but will get the random busy, but not FREEZING!!!!

Thomas-tuta, 2023-04-20:
how did you solve it?

radek8, 2023-04-20:
> how did you solve it?

Try setting a lower communication speed on the serial port

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761#issuecomment-1509852484

benjmharless, 2024-12-04:
I had a similar issue where when printing from a disk on a BTT TFT35 v3 e3 connected to a SKR1.4 turbo.
I fixed it by enabling RX_BUFFER_SIZE and TX_BUFFER SIZE and setting them to the highest option, and by changing the serial ports to match this: https://user-images.githubusercontent.com/54359396/132397091-d596abcf-750f-422a-bb59-afafc246dc58.jpg 
Hope that helps!