bigtreetech/BTT-TFT35-E3-V3.0 #87: TFT appears to communicate with SKR2, then returns to select touch/marlin screen

hikerjohnson, 2022-11-17:
Hello,

I am working with a TFT35 V3 E3 that is connected to an SKR2 Rev B; all on a Creality Ender 3 Max.  I have had this collection working in the past, but have recently redone the firmware to add a second Z servo.

Currently, I appear to have communication between the screen and board, and can select either Marlin or touchscreen mode, and will flash to the appropriate screen for a moment, but then everything returns to the select Marlin/touchscreen mode.

I believe my configurations are all correct per the documentation, and am not sure what to try next.

Please advise; config files are here: 

https://drive.google.com/drive/folders/12XBUJzFbedkjUfIk8iVuxwUQiNC5DoGO?usp=sharing

XeroZar, 2022-11-17:
No worries... So I haven't played with skr2 board yet, but from what I know of the previous ones EXP1 from the board should go to EXP 3 on the LCD. Double check your documentation but that should be the problem.

hikerjohnson, 2022-11-17:
Video of connecting EXP1 from the SKR to EXP3 of the TFT here: https://drive.google.com/file/d/1pthxOioo0t5YJ9O0K90cbX6owy7hX_zG/view?usp=share_link

I tried again, on a whim, lifting the reset pin on the RS232 cable, which made the display do some interesting things, but really I don't think it was a step forward.  Can video if you think its useful.

hikerjohnson, 2022-11-17:
The documentation I've been using is here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#connecting-the-tft-to-the-mainboard

I believe I read the connection table correctly for wiring.

XeroZar, 2022-11-17:
Hmm, connections look fine. Ok just connect the tft connection, remove the 2 serial grey cables and tell me what happens.

hikerjohnson, 2022-11-17:
Identical behavior to the first post - choice screen, flash to something that looks OK, then back to the choice screen.

XeroZar, 2022-11-17:
Did you compile the firmware?

hikerjohnson, 2022-11-17:
I'm not sure how to answer that - I used the marlin builder in VScode to crank out the firmware file.

XeroZar, 2022-11-17:
Was this the original one or did you download from somewhere?

hikerjohnson, 2022-11-17:
Do you mean the firmware for the control board or the screen?  The screen is the latest files from biqu's github repository, and the control board is the latest stable marlin release

XeroZar, 2022-11-17:
Both... ok do you have rev a or b of the skr 2 board... I don't see anything in the picture

hikerjohnson, 2022-11-17:
It's a rev B board with the 429 processor

XeroZar, 2022-11-17:
ok try the official BTT SKR 2 firmware, just use defaults to see if it works... https://github.com/bigtreetech/SKR-2/tree/master/Firmware/Marlin-bugfix-2.0.9.2.x

hikerjohnson, 2022-11-17:
I'll give it a shot, may be a day before I get that done; away from the machine now.  Thank you for the help!

XeroZar, 2022-11-17:
unplug everything as if it was stock, if it works, then build on that. Hope it helps.

radek8, 2022-11-18:
This is a mechanical problem.
The display is bolted to the Ender3 frame
The display touches the frame and brings up a menu.
If you unscrew the display from the frame, does the problem stop?
The washer between the frame and the display will solve it.

hikerjohnson, 2022-11-21:
Radek, My jaw is on the floor.

There is a reset button on the face of the display board that, when the screen is mounted, is just on the very edge of being pressed - as soon as I unscrewed the board, poof, no problem anymore.  The machine seems to respond in Marlin mode!

Unfortunately, it shows no printer attached in touchscreen mode, so there is some troubleshooting to be done there.

Thank you!!!

XeroZar, 2022-11-21:
> Radek, My jaw is on the floor.
>
>
>
> There is a reset button on the face of the display board that, when the screen is mounted, is just on the very edge of being pressed - as soon as I unscrewed the [...]

Wow lol it was that simple... it's weird though it didn't happen before your upgrade...

hikerjohnson, 2022-11-21:
Right?  I may not have had the board mounted then; it was some months ago.

I have now resolved the no printer issue; was just baudrate.  Now going through the regular hell of configuartion finalization.  

Thank you both again!

XeroZar, 2022-11-21:
Happy printing. 😃

radek8, 2022-11-21:
The problem was not with the reset button. This should reset the display and Merlin.
Your problem was the recurring mode selection menu. This means that the frame was pressing on the touch layer of the display.