bigtreetech/BTT-TFT35-E3-V3.0 #90: setup BLtouch and edit mesh error

Blisk, 2023-04-04:
I am trying to setup mesh edit, because now bl touch tries to probe bed out of limit where is nothing.
I tried everything but still stuck on it. Can anyone help me how to setup mesh whre bl touch probe can detect heatbed?
When I press on that yellow dot it tries to probe BLtouch out of heatbed.
Also I get this error when I try to edit mesh.
[image]
[image]
[image]
[image]

Blisk, 2023-04-05:
You mean for display or for motherboard? I assume you think for display in config.ini

XeroZar, 2023-04-05:
On your mother board firmware... I am assume it's Marlin?

Blisk, 2023-04-11:
I did that too, but still can not manage to make printer check all 9 points of heatbed. What I am doing wrong?
Here is video
https://nextcloud.povej.net/index.php/s/LYqaHFxzoypgNgi

XeroZar, 2023-04-11:
> I did that too, but still can not manage to make printer check all 9 points of heatbed. What I am doing wrong?
> 
> Here is video
> 
> https://nextcloud.povej.net/index.php/s/LYqaHFxzoypgNgi
> 
> Also when I press on mesh for PLA I get this error
> 
> 
> 
> [image]
> 
> 

Have you enabled "#define G26_MESH_VALIDATION in Marlin config.h file?

Blisk, 2023-04-11:
Have you enabled "#define G26_MESH_VALIDATION in Marlin config.h file?
Yes I did and uploaded new firmware to printer

Blisk, 2023-04-12:
It is time to replace the display with a regular one before happened more damage to the printer.
[image]

XeroZar, 2023-04-12:
> It is time to replace the display with a regular one before happened more damage to the printer.
> 
> [image]
> 
> 

Did you change the settings that match with your firmware of the main board?

Blisk, 2023-04-12:
Which settings?
Both board and TFT si from bigtreetech. I upload both firmware for ender 3 pro.

radek8, 2023-04-12:
It means that you have the probe offset set incorrectly in Marlin.

radek8, 2023-04-12:
therefore

[image]

radek8, 2023-04-12:
can you send the configuration files?

Blisk, 2023-04-12:
Ok, thank you for heping.
here in zip is Configuration.h from marlin which is on motherboard and config.ini from firmware for TFT35
[marlin.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/11215047/marlin.zip)

radek8, 2023-04-12:
Opravdu si sám kompiluje Marlinj?
Zaslaná konfogurace neodpvídá obrázkům ktreré jsi poslal.

Blisk, 2023-04-12:
I have compiled marlin in visual studio for BTT motherboard.
And for TFT35 I have download precompiled firmware.
marlin bugfix zip file si for motherboard, other zip file is for screen. 
here are both firmwares on this link, it was too big to attach it here.
https://nextcloud.povej.net/index.php/s/P7GwQwfnsDa25Z4

Blisk, 2023-04-17:
Any help here please??

radek8, 2023-04-18:
You have UBL alignment defined in the configuration.
I use bilinear with printing on a mat made of mirror.
  I have no personal experience with UBL.

According to the configuration, you have a defined network of 10x10 points.
In the photo you posted above, there are only 4x4 points.
That's why I'm not sure if you're using the configuration you sent.
I think the 10x10 grid for ender is unnecessarily too dense.
(I can only get by with a 5x5 grid on Ender)
The probe has an offset of 40mm on the X-axis, therefore it can never reach the right edge.
Try to define the edges of the probed network reduced by the offset here.

[image]

radek8, 2023-04-18:
Will you try this?

[image]

radek8, 2023-04-18:
Or this.

[image]

Blisk, 2023-04-18:
thank you for this I almost give up on BTT.
What I upload here is what I use on my ender 3 pro, exactly that. Because I asked for help so it is what is uploaded.
I will try on configuration_adv.h what post and let you know

radek8, 2023-04-18:
If you have a board from BTT or MKS it is not important.
You still have to adjust the configuration yourself.
Downloading something compiled somewhere is not good.

radek8, 2023-04-18:
Don't want to use bilinear leveling?
It works great for me

Blisk, 2023-04-19:
> Don't want to use bilinear leveling? It works great for me

Do you mean this one??
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

Blisk, 2023-04-19:
> Or this.
> 
> [image]

I didnt find this option

radek8, 2023-04-19:
> > Or this.
> > [image]
> 
> I didnt find this option

configuration_adv.h

radek8, 2023-04-19:
> > Don't want to use bilinear leveling? It works great for me
> 
> Do you mean this one?? do i need to setup anything else for setup bilinear? I have disabled mesh and enabled bilinear.
> 
Yes, this one. 
#define AUTO_BED_LEVELING_BILINEAR

+setting a post higher

[image]

+the number of points in the grid, for example (5):

#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)

   // Set the number of grid points per dimension.
   #define GRID_MAX_POINTS_X 5
   #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

Blisk, 2023-04-19:
I did this now and upload firmware but BLtouch probe still goes out of bed limits.
[image]

Blisk, 2023-04-19:
I have enabled all of this and upload new firmware 
[image]
[image]

but still probe goes out of bed limits
[image]

radek8, 2023-04-19:
I asked you to measure the distance of the probe from the nozzle in the X-axis and in the Y-axis.
You didn't answer

radek8, 2023-04-19:
[image]

Blisk, 2023-04-19:
> I asked you to measure the distance of the probe from the nozzle in the X-axis and in the Y-axis. You didn't answer

sorry I missed that
X = 28 mm
Y = 45 mm

radek8, 2023-04-19:
#define NOZZLE_TO_PROBE_OFFSET { -28, -45 -1 }

#if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT 30
  #define PROBING_MARGIN_RIGHT 30
  #define PROBING_MARGIN_FRONT 48
  #define PROBING_MARGIN_BACK 48
#endif

radek8, 2023-04-19:
A poorly defined probe offset was your problem!
That's why UBL didn't work for you either.

radek8, 2023-04-19:
You still have to measure and calibrate the distance of the probe from the nozzle in the Z axis

Blisk, 2023-04-19:
> A poorly defined probe offset was your problem! That's why UBL didn't work for you either.

Ok thank you I manage this now.

Blisk, 2023-04-19:
> You still have to measure and calibrate the distance of the probe from the nozzle in the Z axis

I know this I will do this with baby steps option. Problem is also, because I need to change bed size to smaller, don't know where that option is, didn't find it in settings of printer

radek8, 2023-04-19:
[image]

Blisk, 2023-04-19:
> [image]

thank you

Blisk, 2023-04-19:
One more more thing. This is the same print and BLtouch have probed 16 points before printing, but on the right side filament is smashed and it looks like nozzle too low until on the left side print is ok. As I said, the same print, photo from the left and right side. Why printing is not the same on whole bed? 
[image]
[image]

radek8, 2023-04-19:
Sometimes I had that problem too.
There is a big difference if you level a cold bed and then press on a heated one. The bed is deformed by the heat.

I opened an issue ticket on Marlin's Github,
that there is a problem with the leveled bed in Bilinear mode.
They replied that the leveling calculation is fine. The problem was apparently in the mechanics of my printer. Clearance in bearings, mounts, motor arms, etc.
That's why I redesigned the printer (also Ender3).
I added another trapezoidal screw on the right side.
I put the trapezoidal screws of the Z axis with a half pitch and locked the screws at the top and connected them with a belt.
He replaced a more powerful motor on the Z-axis.
I put a glass mirror on the bed and printed it on a glass base, which is much straighter than the original plastic one. The underside of the print is then beautifully smooth.

If I want to flatten the pad, I heat the pad first. I will level the corners via the touchscreen menu (Level Corner) to an accuracy of +- 0.05mm.
Then i start the leveling.
If it seems to me that one side is a bit tight, I turn the screws under the bed on this side by approx. 10-20° and it's fine.
After handling the pad for some time, I level the bed again.
But it is not needed before every print.

So check that the drives and bearings are stiff enough...

Sorry for my English, I'm using Google :-)

Blisk, 2023-04-19:
thank you for your time to help I really appreciate that I almost give up on BTT. they have very lame support.
your native language is close to mine and I was in your country and I understand mostly of your language.
can you put some pictures of what you are describing it helps alot.
mostly printer starts to work and I can print on it that is important this now is just polishing to make it perfect.
thank you again.

radek8, 2023-04-20:
[image]
[image]
[image]
[image]

Blisk, 2023-04-20:
thank you
I have for Z axis 2 motors.

radek8, 2023-04-20:
For Z-axis threaded rods, I recommend replacing the thread with a smaller pitch.
The original Ender bar is too big a climb.
I used a bar with a half pitch as opposed to the original.
You get 2x more motor power at the same current and 2x higher accuracy.

Also throw away the ruined bed mat and use the mirror.
A smooth surface on the product is better.
To improve adhesion to glass I use:
for PET-G - Hairspray. The cheapest one without vitamins and perfume hardness 5
for UTP, PLA and ABS - 3Dlack spray.
Since I've been using it, I haven't had a problem with the first coat sticking.

radek8, 2023-04-20:
I bought this
length 400mm pitch 4mm (Lead 4mm)
https://www.aliexpress.com/item/32818374370.html?spm=a2g0o.cart.0.0.531c38daUUKOKG&mp=1

New on the left, Ender on the right
Originally it was 8mm for one revolution of the engine,
now I have 4mm on the Z axis per revolution

[image]

I'll think about this

https://www.aliexpress.com/item/32808431280.html?af=4043356_1&cv=39374878&cn=42rtfenagh1nns1n2y31zqnz5k72ft67&dp=v5_42rtfenagh1nns1n2y31zqnz5k72ft67&product_id=32808431280&afref=https%3A%2F%2Falipromo.com%2Fredirect%2Fcpa%2Fo%2Fqgrgvysgbby0grndguqqqhkw573mi25v%2F&aff_fcid=70f941e3f3e74cfdbef45eed5cc0f485-1682014439057-08653-_labmVt&tt=API&aff_fsk=_labmVt&aff_platform=api-new-link-generate&sk=_labmVt&aff_trace_key=70f941e3f3e74cfdbef45eed5cc0f485-1682014439057-08653-_labmVt&terminal_id=290428c8106c48438ea1ffecefca338b&afSmartRedirect=y

Blisk, 2023-04-20:
I was searhing for 4mm pitch but i found only 8mm or 2mm. link you posted is for brass nut.
https://www.aliexpress.com/item/4001332632943.html?spm=a2g0o.productlist.main.11.5ee3af02Se3t5K&algo_pvid=7f83d457-8fb3-42d5-aa3e-99541e50d1a1&aem_p4p_detail=202304201212543539437819070470001593876&algo_exp_id=7f83d457-8fb3-42d5-aa3e-99541e50d1a1-5&pdp_npi=3%40dis%21USD%217.31%216.58%21%21%21%21%21%40211beca116820179742857556d0763%2112000018205142658%21sea%21SI%21138963833&curPageLogUid=BzzJ2ei9NnYJ&ad_pvid=202304201212543539437819070470001593876_6&ad_pvid=202304201212543539437819070470001593876_6

radek8, 2023-04-20:
It will have a stroke of 2 mm per revolution.
That will probably work too :-)

radek8, 2023-04-20:
Yes, the reference was to the matrix.
It's a forged nut with a spring that should reduce the play on the thread...

radek8, 2023-04-20:
the threaded rod link was here:
https://www.aliexpress.com/item/32818374370.html?spm=a2g0o.cart.0.0.531c38daUUKOKG&mp=1