bigtreetech/BTT-TFT35-E3-V3.0 #91: BTT SKR MINI E3 V3 AND TFT35 E3 V3.01 TOUCH MODE CONSTANTLY RESETS AND 'NO PRINTER ATTACHED' MESSAGE IS CONTINUOUS.

Jhendrix0875, 2023-06-30:
I HAVE AN ENDER 3 PRO AND I JUST REPLACED THE MOTHERBOARD AND THE SCREEN. I UPDATED THE FIRMWARE TO THE LATEST ON THE TFT DEVICE. I HAVE CHECKED 50 TIMES TO MAKE SURE THAT THE SERIAL CABLE IS PLUGGED IN CORRECTLY, AND I HAVE VERIFIED THE BAUD RATE WITH THE PRINTER AND THE TFT ARE THE SAME. IT WORKS IN MARLIN MODE AND I AM ABLE TO CONTROL THE PRINTER THAT WAY. HOW DO I GET THE TOUCH MODE TO WORK?

Jhendrix0875, 2023-07-02:
Here are my config and config adv files
[Configuration.zip](https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/files/11931263/Configuration.zip)

radek8, 2023-07-04:
Close the ticket if you have solved the problem. Thank you

radek8, 2023-08-14:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2811