bigtreetech/BTT-TFT35-E3-V3.0 #92: SKR Mini E3 V3.0 and TFT35 E3 V3.0.1

Artimus70, 2023-08-21:
Hello all,
this is my first post on this platform and i hope you guys may help me in understanding the issues i'm having.

I bought an SKR mini e3 v 3.0 and its 3.5 tft touch display (3.0.1) to revive my Anycubic i3 mega since the original MB is gone (its just a batch of motors and microswitches after all)

I had to recompile the fw for the board and i did it starting from Marlin 2.1 + bugfixes. (also the configurations are bugfix of course)
I updated the LCD firmware (recompiling with the latest updates released by BTT).

The LCD can control the mainboard when in Marlin Mode doing all the right stuff (homing, heating, moving around) but in Touch Mode it reports "No Printer Connected".
No matter which baudrate i set.
So i checked the cables, the connections, the serial ports declared on both firmwares (LCD and MB)... but no luck.
I then tried different modifications to the configuration.h... new compile... but no luck

The only way to see the printer connected to the LCD in Touch Mode is to connect a raspberry pi with octoprint via usb to the board. At that moment i can see the printer attached to the display but i cannot control it from the lcd. (no...its not in listening mode)
It seems the serial port for the LCD is alive only when talking via usb.

Any suggestion is well appreciated.

Thanks in advance to any of you spending time to help.
Have a great day.

radek8, 2023-08-29:
It looks like a corrupted serial port pin. It has already been addressed here several times.

radek8, 2023-08-29:
https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

Artimus70, 2023-08-30:
Many thanks Radek8.
I'll try the solution proposed and i'll let you know. 
Have a nice day.

Artimus70, 2023-09-01:
Hello Radek,
thanks a lot for your suggestion.
As you may expect it worked perfectly.
I recompiled the TFT FW declaring the serial port3 as printer port, splitted the rs232 cable in two, added a new jst connector and plugged in serial3 of the display.

I didn't expect this to be a so frequent issue on this kind of board.
For sure there is some kind of design issue.
I verified that the serial lines (TFT side) are short circuited for some reason.

Thanks for you support.
Have anice day.

radek8, 2023-09-01:
You're right, it's a fairly common problem.

radek8, 2023-09-01:
If you have resolved the issue, please close the ticket

petarch96, 2024-03-31:
Artimus70, I am just starting this complex rebuild for me as well and I am having some issues with installing fw and all of that work after wiring everything. is it possible for you to maybe give me a tip or two?

Artimus70, 2024-04-01:
> Artimus70, I am just starting this complex rebuild for me as well and I am having some issues with installing fw and all of that work after wiring everything. is it possible for you to maybe give me a tip or two?

Hello Petarch96,
Explain your problems, please, and i'll try to help.
Its been a while since I made it working but i'm sure we will figure out what you need.

petarch96, 2024-04-02:
Thank you for answering on such short notice. Well currently I have more or less wired a lot of things but still missing the knowledge to handle all variables. Regardless the setup:
Anycubic i3 mega/ mega-s
MB: skr e3 v3 mini
Screen: TFT 35 also from BTT

Problem description:

> Wiring of end switches due to the removal of the i3 mega hub pcb: When I cut the hub connector do I need to split the ground cable into 5 grounds in order to connect a signal and a ground to each end switch? (PS: I can attach a screenshot later of how I have planned on doing that part).

> firmware: I am not familiar with firmware installations  and I'm having trouble finding the correct one. I'm not sure if I need an exact firmware for this setup or if I'm good with using something else(like ender 3)? Do I need to manually adjust firmware according to setup and if yes how does that happen?

> bl touch: I have also purchased a BL touch in order to have better leveling but I'm unsure how the wiring should be done as some people on Enders attach 2 pins to z-switch or all 5 to the BL touch connector.  How would that work if I still have one cable from the i3 mega hub which is supposed to be connected to a BL touch pin...

I apologize for the many very open ended questions and not so much concrete small issues. It's my first time doing any PCB and wiring work. I would greatly appreciate your help (here or private messages if more suitable for the amount of information). I would be happy to add the information to your "ticket" so other struggling enthusiasts can benefit.

Have a great day!

Artimus70, 2024-04-02:
GoodMorning Petarch96,
as i mentioned it's a long time i don't deal with SKR..
Anyway let's try to figure out few things. I also hope someone from this group will get-in giving an helping hand.

1. Each end switch needs two wires to make it working. 1 for signal and 1 for, usually, ground. -> so the answer is yes.

2. For the firmware try the Stock Marlkin for ender3 (should work) if it doesn't go to step 3.

3. Build your own FW:
 a. Download and install Visual Studio Code (its free) https://code.visualstudio.com/download
 b. Download the Marlin Source code for your board (please check the SoC model you have) https://github.com/bigtreetech/Marlin/tree/SKR-mini-E3-F401
c. do all the adjustments you may need. -> Compile -> flash the board.

4. I'm not familiar with BLTouch but i can tell you what i would do in your shoes: first make the printer work with the microswitches and later try to figure out out to make use of the BLtouch.

5. Be aware that the display "could have" a little problem with the default serial port. You have to change it in the TFT firmware in order to use a different port (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware) and wire the serial port from the SKR to the TFT accordingly. Again try to use the display in marlin mode first to be sure that the printer is working as expected and later try to fix (if needed) the tft serial port issue. (Please check your TFT display board revision and adjust the config.ini file see example here attached)
[image]

There's plenty of tutorials on how to compile and flash the boards but in case you need just drop a message here.

I hope this helped a bit.
Have a nice day.

radek8, 2024-04-02:
@petarch96 
Your configuration is not very standard and it will be difficult to search. compiled FW that would match your configuration.
So you have to compile your own FW.
You can connect the whole BL-touch to the connector for the probe, or split it and connect two pins to the Z-stop connector. But it always depends on what is configured in the FW.

radek8, 2024-04-02:
All limit switches can only have one grounding pin and it doesn't matter which pin (X,Y,Z) it is terminated on.
Attention, do not confuse the ground with a fan or heating element!!! These must be conducted separately and must not be connected.
For BL touch you need to stretch another wire at least for the servo pin and the 5V power supply.

petarch96, 2024-04-02:
1. @Artimus70 @radek8 - this is the way I had planned on doing the wiring for my setup:
[image] (2 representations because it would have gotten very hard to see). According to @Artimus70 i need to split the black cable (Nr. 6 / Top picture) into 5 different grounds and connect them to the corresponding grounds for each signal. According to @radek8  i can ground the Nr.6 to an end switch X or Y or Z and it would be fine. So far not sure which one it would be haha. I have yet to buy a wago splicing connector if I would go with Artimus70's tip.

2. I will try the Ender 3 config and hope it would work. Not yet sure how I would verify this to be honest. LIke, would the motors and endswitches not work and other components like heaters and thermistors or would I just get errors on the screen?

I found this in the manual of the motherboard: 
[image]

Currently I dont have the endswitches connected but through the TFT35 I did manage to move the axis. I suppose this is good news. My concern (if I would be able to continue with this standard firmware) is that I have 2 Z Axis motors and I think the Ender 3 uses only 1 so I'm not sure how I would activate another motor to be controlled by the motherboard

Regarding downloading the Firmware I'm not even sure which one of those 2 I have.
[image]

also which file would I need to exactly download.

On the page of MB I found this link here which redirects straight to Ender 3 Config files: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin

5. Have not yet gotten far but I will try to watch some videos and hopefully I can learn how to configure the firmware

radek8, 2024-04-03:
Your printer has 2 extruders and 2 Z motors?

petarch96, 2024-04-03:
No - 1 extruder, 2 Z Motors

radek8, 2024-04-03:
So why are the connectors on the i3 mega hub labeled E0 and E1?

If you only use 1 extruder and 2 Z motors, you have to synchronize the motors with a toothed belt. 
The SKR E3 mini has only one motor controller, therefore the sum of the currents of both motors must not exceed the permitted current of the controller.
Then you will define FW as if you had only one Z motor but with twice the current.

radek8, 2024-04-03:
To your question about the difference between SKR mini F401 and SKR mini V3.0 G0B1
The difference is in the processor used.
The first uses an STM32F401 processor, the second G0...

petarch96, 2024-04-03:
> So why are the connectors on the i3 mega hub labeled E0 and E1?
>
> If you only use 1 extruder and 2 Z motors, you have to synchronize the motors with a toothed belt. The SKR E3 mini has only one motor [...]

This is in the specifications: 
[image]
So I guess I have the G0 processor. 

> So why are the connectors on the i3 mega hub labeled E0 and E1?
>
> If you only use 1 extruder and 2 Z motors, you have to synchronize the motors with a toothed belt. The SKR E3 mini has only one motor [...]

In the original AnyCubic configuration it was the same. I had one extruder and 2 Z Motors. I guess what the firmware does is control the seocnd Z-Motor through the E1 port. E0 is for the extruder motor. I dont understand how the SKR MB has only one motor controller when it specifies ZAM and ZBM. Those 2 should be separate motor controllers if my logic isnt faulty here. 

[image]

The question here would be in my opinion. How would I allocate the right signals to the correct motor (same with the Z-End switches)  through the firmware. 

PS: I have yet to open the Ender 3 Firmware is Visual Studio and see what awaits my few brain cells :D

radek8, 2024-04-03:
To be sure which version you have, take a physical look at the processor. The sellers on the site sell different veres and don't really care about the specifications.

If it is about two Z motors, then your SKR E3 mini Board has only 4 motor controllers (X,Y, Z, E0), there is no other one.

[image]

Connectors ZA and ZB are connected in parallel to one controller.
If you want to control two Z motors independently, you bought the wrong board.
The only way to control two Z motors with this SKR E3 mini board is to synchronize them with a belt. on the Z controller, you can set double the current, but you must not exceed the maximum allowed for the given controller.

petarch96, 2024-04-03:
Well this just worsened my situation by a whole lot haha. That means that I need to do hardware upgrades to the printer aswell... :D

petarch96, 2024-04-03:
PS to last comment: https://drive.google.com/file/d/1DTscTKrLy0et0B8UPKGcx3DHsZTG1HCL/view?usp=drive_link a little video of showing that both motors are rotating

radek8, 2024-04-04:
Your SKR E3 mini board has drivers connected in UART mode. So the current and the number of microsteps are set in the firmware before compilation.
Current settings can also be changed later from the screen menu. The condition is that the motor drivers were set in UART mode in the configuration before compilation.
That's why it's important to compile your own FW so you know what you're up to. You can also set the current reduction on the Z-axis controller if the axis is not moving. This can greatly reduce the temperature of the motor for long prints while still having enough motor power to move the ramp.

Yes, the motors both spin at the same time on the same signal. the problem is that if you deactivate the motors or turn off the printer, the motors can turn against each other and the ramp that the extruder travels on will tilt. Both the ramp and the pad are calibrated to hundredths of mm. Therefore, to prevent this, there are two options. The first one is that each motor has its own controller and when switched on, the ramp is leveled electronically using BL-Touch or limit switches. But you don't have that. You would have to buy another control board.
The second option is to synchronize the motors mechanically - with a belt.

radek8, 2024-04-04:
Attention. During your experiments, never connect or disconnect the motors while the printer is on.
The motor controller may be destroyed.