bigtreetech/BTT-TFT35-E3-V3.0 #100: displayTFT35 E3 V3 and klipper

Jeff2405, 2024-09-19:
I just bought an SKR1.4 card and a TFT35 E3 screen.
I connected the screen with both EXP1 and EXP2 cables as well as the RS 232.
When I start the screen lights up, works in touch but rings continuously when I switch to Marlin mode.
I have a "no printer attached" message. I changed the speed of transmission and it's the same.
I added in the CFg printer:
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>

dkarpes, 2024-09-19:
Hi 

did you plug it in the last most right (exp3 i think it is) and you need to bridge it with this code

### BigTreeTech TFT TouchScreen emulated 12864 mode: Set the `display` in `printer.cfg` to the following parameters
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: PD8 # status led, Virtual MISO
spi_software_mosi_pin: PD6
spi_software_sclk_pin: PB9
en_pin: PB8
encoder_pins: ^PA10, ^PA9
click_pin: ^!PA15

[output_pin beeper]
pin: PB5

Jeff2405, 2024-09-20:
no I connected exp1 and exp2 as well as RS232 on the screen with TFt on the card.
I will try your solution

Jeff2405, 2024-09-20:
hello
I get this error message "Pin 'PB8' is not a valid pin name on mcu 'mcu'" when I put this config

dkarpes, 2024-09-21:
I thought you had a BTT skr mini e3 v3.0. But you need to translate the pin names from my code to your card
But that is the LED you can try to piut a # in front of it to disable the led, and try if that works.
As far i could find my PD8 is your P0.17
check and translate my pins to your board:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4-SCH.pdf

dkarpes, 2024-09-21:
Found this somewere on a form:

Pin Mapping Summary for SKR V1.4:

SPI Software MISO Pin: Replace with PA12 or similar GPIO.
SPI Software MOSI Pin: Replace with PA15 or PB14.
SPI Software SCLK Pin: Replace with PB10 or PB3.
EN Pin: PB8 (usually directly available).
Encoder Pins: PA10, PA9 (should be directly usable).
Click Pin: PA15 or PC13.

Always double-check your specific SKR V1.4 board documentation, 
as some pins might be tied to particular functions depending on the firmware configuration.

maybe you can do something with this!

good luck!

Jeff2405, 2024-09-24:
Hello

I don't live on the document where you see the pinout numbers (PA12, PA15, etc..)

dkarpes, 2024-10-01:
sorry, didnt see your message:
found this in a marlin pin mapping file
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Firmware/Marlin-bugfix-2.0.x-SKR-V1.4/Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_4.h

maybe you can translate the pins to your display